makers-for-life/makair #3: Choosing the pressure sensor

Benhalor, 2020-03-19:
Hello,

Congratulations on your project. I'm impressed with how fast you're moving!
You said in the video that the pressure would be monitored by a pressure sensor. 

If you need help for choosing the pressure sensor, you can contact me : +33 (0)695358125

Something like this one could be great :
https://fr.farnell.com/honeywell/abpdant005pgaa5/capt-pression-5psi-axial-avec/dp/2490459

0 to 5 PSI sensor, which means 0 to 258,5 mmgh
Absolute Precision : +-1.5%FSS which means +- 0.075 PSI or +-3.9 mmgh
Relative Precision : 0.25%FSS which means +-0.0125 PSI or +-0.64 mmgh
5V sensor, because the arduino uno that you choose is 5V.
through-Hole sensor : easier to solder
14€

Gabriel

waxzce, 2020-03-19:
For the reccord, we did have a call with @Benhalor and solve many problmes thax to him

pi-r-p, 2020-03-20:
Hello, 
I went through the electronic part (respi_V0.fzz) ( @hbs send me a link to the project) , same remarks... In french:

Le capteur MPX10GP n'est pas le meilleur. La plage est bonne (10kPa pour un respirateur, ça va, la pression usuelle montera au grand max à 5kPa ([d'après une doc médicale ici](https://www.srlf.org/wp-content/uploads/2020/03/2020-FP-Respi-3-Principes-ge%CC%81ne%CC%81raux-de-re%CC%81glage-des-modes-contro%CC%82le%CC%81s-et-assiste%CC%81s.pdf)), mais il ne faut pas prendre un modèle avec une sortie différentielle derrière, et encore moins mettre un AOP comme celui là derrière (lm741, ça s'alimente en +/- 15v si vous voulez en tirer qqch, et il faut gérer l'aspect différentiel). 

Nxp fait aussi des modèles plus récents avec électronique intégrée, qui te font une belle sortie rail to rail 0-5V. Typiquement https://www.nxp.com/docs/en/data-sheet/MPX5010.pdf , c'est déjà plus simple à cabler sur un arduino. Prendre un modèle avec tuyaux, dont un reste en l'air à la pression ambiante, typiquement MPX5010DP. L'amplification n'est plus nécessaire, on peut se contenter de baisser la ref anologique de l'ADC (AREF sur un arduino). 

Attention, il faut aussi créer une bonne référence de tension, et isoler la partie capteur/ADC du reste pour pas avoir un signal trop bruité. Sans séparation puissance/controle, le 5V risque de fluctuer à chaque mouvement des servomoteurs, et fausser la mesure de pression. Donc:
- on alimente en 9 ou 12volts
- on met derrière un régulateur 5V à découpage pour l'arduino et les servos (n'importe quel buck 5v trouvé sur le net)
- un régulateur 5v linéaire (un 7805) pour le capteur, avec quelques condos de découplage (470uF derrière le 7805, 100nF au plus proche de l'alim du capteur, soudé dessus de préférence si il n'est pas sur un circuit imprimé). 
- un régulateur 3.3v linéaire sur AREF (3.3v expoiterait mieux la dynamique de l'adc), pareil, avec des condos de découplage.

Et si c'est un capteur Nxp de cette série il faut une procédure de calibration initiale, car c'est pas un capteur super précis. Le capteur Honeywell suggéré par @Benhalor est deux fois plus cher, mais 10 fois plus précis... Mieux vaut prendre celui là !!

À confirmer par un médecin, mais la pression minimale à maintenir en fin d'expiration est importante aussi, d'après le doc dont j'ai mis le lien plus haut.

Pepeuch, 2020-03-21:
Le prix d’un capteur même 10X plus cher est toujours mieux qu’une vie, le problème va être la livraison pour que cela arrive au plus vite, avec les mesures de confinements et donc de livraisons va être un nouveau dileme.

Mefl, 2020-03-24:
@Benhalor a été intégré au slack

closed 2020-03-24 by Mefl