makers-for-life/makair #4: Forms and maps to prepare for contributors willing to help for mass production

Shalf, 2020-03-19:
Why?
Work from a precise cartography of shops/venues, people (and skills), machines, and stocked supplies.

How?
Trough a form, or anything pre-existing that could be remixed and improved.

[This form](https://ventilator.herokuapp.com/) is being shared by the UK gov, and is a good inspiration for what ours could look like.

An AMI (Appel à Manifestation d'Intérêt) is being held by the EU Commission, with French relays, to do a census of companies with 3D printing production capacities willing to help. We took contact.
We are working wth the RFF (Réseau Français des Fablabs) to find a local cartography of this distributed network alredy engaged and willing to help.

valeriansaliou, 2020-03-19:
Excellent, thanks for the information & contact initiative w/ the AMI system.

Shalf, 2020-03-23:
From CS RFFlabs : French makers' data will end up here https://github.com/co-demos/covid-viz/