makers-for-life/makair #10: How does a layman on the subject that has a 3D print can help?

renata-castro, 2020-03-21:
Hello.
today I received the link to this website by a co-worker who is a coder and in our company we have a 3D printer. we would like to help but we want to know if hospitals will accept these parts, if there are no sterilization requirements that would prevent us from providing, in addition to other possible issues. is there an approval from the WHO? what do you recommend before starting the production of the parts? 
thank you.

valeriansaliou, 2020-03-21:
Hello there,

The parts are not yet finalized; we’ll make an announcement whenever the tests were done and that the parts are ready for production.

Regarding the printing material, it has to be as chemically stable as possible, so we’re looking for medical-grade ABS (neutral color), or even PETG.

tuxinoo, 2020-03-21:
PETG is the better choice because it’s a material stronger and safer (ie water bottles and for food) than ABS (toxicity). This project is dealing with patient’s health (safe) and mecanical movements (stronger) so I recommend PETG.
Nylon is also a choice but not all 3D printers are capable to print with.

valeriansaliou, 2020-03-21:
@tuxinoo we have a MakerBot Method at Crisp; a Nylon spool order is on the way, should be received on Monday or Tuesday ;)

tuxinoo, 2020-03-21:
Nylon is a better choice if you have a good 3D printer. ). I have just an Ender 3 at home.

renata-castro, 2020-03-21:
Thank you for the clarification. I've seen an article today that reinforces the need for this:
https://www.3dprintingmedia.network/covid-19-3d-printed-valve-for-reanimation-device/?fbclid=IwAR2gdR9bdaRVP9h7n1xUMNd4cqFSX3PaTXNwH9Y6qe8AMNdxp0OHJ-cnzYE

renata-castro, 2020-03-21:
I'm in contact with a former owner of a hospital in Rio de Janeiro and also trying to get a closer contact to head doctors here in Oregon and California. 
How are you guys planning to distribute and proceed with the sterilization of the pieces?

fr33z00, 2020-03-22:
Silicon is very neutral, as far as I know.
There are two ways of using it: 
- as a cover on any plastic (even PLA, then)
- molded, the mold behing realised in any plastic.

Why-Ty, 2020-03-25:
Bonjour à tous, mon entreprise a une formlab 2 a st Étienne de Montluc et une ultimaker 2 extented+. Nous avons quelques bacs de résine pour la formlab et un peu de PETG (BLEU). Sinon, de l'ABS et pla bien sur. Dites moi si vous avez besoin, je ne suis pas sur place mais je peux voir avec mes collegues. Si vous souhaitez imprimer en PEHD, c'est assez complexe mais j'ai des contact en France qui font ça très bien et j'ai un plateau spécifique pour l'adhérence. A dispo pour toutes infos. Yo

dolmen, 2020-03-31:
See also this project from MIT : https://e-vent.mit.edu/