makers-for-life/makair #12: How can I contribute?

alencarfff, 2020-03-21:
Hello, I'm Felipe, I'm a Brazilian experienced web developer and, I found your project looking for a way to helping against the COVID-19 epidemy. Your project is fantastic and I hope all it works well when finalized. I can help voluntarily to build a website to better promote your project and bring more people to contribute to it. Beyond it, I can help to translate part of the project documentation to PT-BR too.

waxzce, 2020-03-21:
I focus on the prototype delivery, I try to find someone to coordinate with you

alencarfff, 2020-03-22:
Okay. I'll wait for your contact.

HugoGresse, 2020-03-27:
French web dev here
Let us know if you need a build a landing page for MakAir or anything front-related.

eliottvincent, 2020-03-27:
Hello @HugoGresse, I can invite you to our Slack, we have a team already working on the front, I think your help would be appreciated. You can send me a mail at dev at eliottvincent dot com

fareivas, 2020-03-27:
Hello, I am a Braziliam experienced with 3D printing for robotics and firmware development. I am trying to replicate/build your prototipe project here in my city. 
1. Can you provide a bill of materials used so far ?
2. How can I help with my skills?

Best Regards
Fábio

JulienCalvignac, 2020-04-21:
Hello all,

First of all, congratulations for such achievements in such short time!
How can I help  as a volunteer ?

I am a systems engineer/architect with strong background in system architecture (safety), controls, electronics, electricity, embedded software and project management.
I am very familiar with agile and CI development, I can help on such topic as well. I can help also on safety and regulatory compliance.

Let me know !! 
All the best !

Julien Calvignac

NopMicrowave, 2020-04-25:
Hi,

I have seen that you plan to use LoRa Radio scheme (in the telemetry, or wherever you want).
I have more than 10 years of experience in LoRa, LoRaWAN in industrial, city, home, on premise, on Raspberry Pi, on private VPS, ...
If you need some help, you can ask, I will help.

Nop

Nopmicrowave.io

ilyautov, 2020-05-09:
Hello, i'm Russian back-end developer, i have strong experience in node.js, golang and if that reasonable in rust, previously i have some hardware projects and i think  could help
I'm want to help project, may be you have any tasks for me?

BSte77, 2020-05-14:
Hi,
I am from Paris, I can help with 10 years background in robotics systems and custom machines, and 10 years in Li-Ion batteries. 

Wining space in the enclosure by exchanging lead-acid batteries by Li-ion one  could help to win space in order to embed O2 bottle and make the makair portable for medical trips.

Good luck!
Bertrand

eliottvincent, 2020-05-15:
Hello @BSte77 , Li-ion cells is something we would like to work on. I can invite you to our Slack if you provide me your email :)

BSte77, 2020-05-15:
Hi Elliot,
Please find here my email and welcome to your inquiries. Really motivated
to contribute to that project!

I am really new in GitHub with really no knowledge on how to do
collaborative work with this plateform.

Looking forward your return and your advices in order to be the most
efficient as possible 😉

As a first exchange, could you tell me more about :
Voltage, I saw 2 batteries.  Are they connected in series or parallel ?
Maximun Instantaneous power needed
Current avalaible capacity
Environement temperature

As you know, there are some Li-ion batteries avalaible on the market, for
exemple for Ebikes that may ne used.

Bertrand.

BSte77, 2020-05-15:
Hi again Elliot,
I answered directly to your email. Also, did you receive my email address or do I have to share it here ??

eliottvincent, 2020-05-15:
@BSte77 Thanks, you may share it here or send me an e-mail at dev[at]eliottvincent[dot]com if you don't want to display it publicly.