makers-for-life/makair #28: [Soft] be sure that every   pinMode(PIN_SERVO_PATIENT, OUTPUT); is OK

waxzce, 2020-03-25:


jabby, 2020-03-25:
WIP here.
https://github.com/covid-response-projects/covid-respirator/pull/21

dsferruzza, 2020-03-25:
I believe this is already handled by the Servo lib we use? → https://github.com/arduino-libraries/Servo/blob/master/src/avr/Servo.cpp#L241

jabby, 2020-03-25:
I close the issue as it is handled by Servo lib.
Feel free to reopen if needed.

waxzce, 2020-03-25:
but not for every pin, whiuch was my point

dsferruzza, 2020-03-25:
I'm not sure to get your point actually... Both servos are OUTPUT thanks to the lib, right? Do we need more?

jtanguy, 2020-03-26:
Yes, for the buttons, pressure sensor and buzzer. 
There is no pinMode set for any pin in the master branch atm

jabby, 2020-03-26:
Ok I'm on it.

jabby, 2020-03-26:
* Servo: :ok: thanks to the [lib](https://github.com/arduino-libraries/Servo/blob/master/src/avr/Servo.cpp#L241)
* Buttons: :ok: thanks to the lib ([here](https://github.com/rlogiacco/AnalogButtons/blob/master/AnalogButtons.h#L105) and [here](https://github.com/rlogiacco/AnalogButtons/blob/master/AnalogButtons.cpp#L34)) and to our code [here](https://github.com/covid-response-projects/covid-respirator/blob/master/src/arduino/respi_v0.cpp#L188)
* Pressure sensor: missing. `analogRead` does not set the pinMode to `INPUT`.
* Buzzer: not implemented yet.

commit bb049888 (2020-03-26): Merge pull request #33 from covid-response-projects/issue-28

Fix #28 Add missing pinMode
closed 2020-03-26 by Mefl