makers-for-life/makair #50: Add a licence to each source file

Mefl, 2020-03-29:


jlbirccyn, 2020-03-30:
Done with e412d7a465cec8bb79fc1c047b9ccc1bf323c1b0

dsferruzza, 2020-04-06:
I'm reopening this because:

1. Headers are not in the current version
2. I believe we did not actually choose a specific license yet? (cc @valeriansaliou )

valeriansaliou, 2020-04-06:
The current general license is the following one: https://github.com/covid-response-projects/covid-respirator/blob/master/LICENSE.md

Question is as follows: do we accept commercial use of our project? If we want industrials to be able to mass produce our respirator, this should probably be allowed, as long as changes on our CAD models, software code, and more are re-licensed as OSS / same license.

closed 2020-05-16 by dsferruzza