makers-for-life/makair #55: Analyse risque : RCM-SW-5

Mefl, 2020-03-29:
The software is validated

cmdm, 2020-03-31:
Validation shall pertain to the application IEC 62304 requirements by having
-architecture
-if possible detailed design, like doxigen generated documents
-unit tests results logs with code coverage rationale
-static analysis logs
-integration and functional tests
-residual bugs recorded in bug tracker

dsferruzza, 2020-04-06:
Current status:

- [ ] **architecture documentation:** we are waiting for things to stabilize
- [x] **Doxygen documentation:** automatically generated and up-to-date (we will do our best to keep them up-to-date as we land evolutions)
- [ ] **unit tests:** not started yet (I'd like to make it happen before the end of the week)
- [ ] **integration and functional tests:** not started yet
- [ ] **residual bugs recorded in bug tracker:** in progress

closed 2020-05-16 by dsferruzza