makers-for-life/makair #107: Software: implement alarms

csommelet, 2020-04-06:
Alarms logic is currently missing.

Related issues describing alarms individually (alarms definition is still work in progress):
 - RCM-SW-19 - Low plateau pressure - https://github.com/covid-response-projects/covid-respirator/issues/99
 - RCM-SW-18 - High plateau pressure - https://github.com/covid-response-projects/covid-respirator/issues/98
 - RCM-SW-14 - Plateau not reached - https://github.com/covid-response-projects/covid-respirator/issues/64
 - RCM-SW-15 - PEEP not reached at 3rd cycle - https://github.com/covid-response-projects/covid-respirator/issues/65
 - RCM-SW-3 - PEEP not reached at 4th cycle - https://github.com/covid-response-projects/covid-respirator/issues/53
 - RCM-SW-8 - Pressure out of range - https://github.com/covid-response-projects/covid-respirator/issues/58
 - RCM-SW-2 - Low Pressure - https://github.com/covid-response-projects/covid-respirator/issues/52
 - RCM-SW-1 - High Pressure - https://github.com/covid-response-projects/covid-respirator/issues/51
 - RCM-SW-16 - Mains disconnected - https://github.com/covid-response-projects/covid-respirator/issues/70
 - RCM-SW-11 - Low battery - https://github.com/covid-response-projects/covid-respirator/issues/63
 - RCM-SW-12 - Very Low battery - https://github.com/covid-response-projects/covid-respirator/issues/59

Alarming system behavior:
- RCM-SW-13 - Maintain PEEP during high prio alarm - https://github.com/covid-response-projects/covid-respirator/issues/62
- RCM-SW-10 - 'Silence Alarm' button should reset 2min timer - https://github.com/covid-response-projects/covid-respirator/issues/61
- RCM-SW-9 - New alarm should cancel Mute - https://github.com/covid-response-projects/covid-respirator/issues/60

Test benches protocols:
https://docs.google.com/document/d/1A3L-puGiDQOrkns229-GeCmOXisWVPol59NXHKI1Foo/edit

dsferruzza, 2020-04-06:
_I'll try to sum up my understanding._

Alarms have:

- **a trigger predicate:** it can depend on various parameters, and may need an internal counter of the number of cycles in a row the predicate was true along with a threshold that will raise an alarm when exceeded
- **an error code** (unique)
- **a severity:** low, medium, high

When an alarm is triggered it has effects depending on its severity:

- buzzer (already implemented)
- LEDs
- display error codes on the third line of the LCD screen (using horizontal auto-scrolling?)

When several alarms are triggered at the same time, only the alarms of the higher triggered severity have effects.

If a user press the _alarm OFF_ button, it snoozes the buzzer for 2 minutes. If an alarm of higher severity is then triggered, snoozing is interrupted and buzzer rings.

If an alarm's predicate becomes false, it is not to be considered active anymore. Effects can continue to happen according to the higher severity of the remaining active alarms, or stop if there are no remaining active alarms.

---

I believe we need some kind of alarm controller (singleton?) that could:

- be called if the predicate of an alarm is true (this might trigger effects)
- be called if the predicate of an alarm is false (this might stop effects)

It would then keep track and handle:

- inner state of alarms (number of cycles in a row it was triggered)
- active alarms
- snooze status (if snoozed, we need to keep track of the timing and the severity that was snoozed)
- LEDs effects
- LCD display

RomainBlanc, 2020-04-07:
When an alarm is triggered, the alarms number will be displayed on the third line of the screen:

[image]

RomainBlanc, 2020-04-07:
[image]
French version

Mefl, 2020-04-07:
due to #120, we need to store the cycle number for each alarm triggered to be able to send the new alarms (ie triggered in the last complete cycle)

hbs, 2020-04-14:
Alarms should trigger telemetry events, so should the user actions.

closed 2020-05-16 by dsferruzza