makers-for-life/makair #120: telemetry protocol

Mefl, 2020-04-07:
The V1 card is connected to a second device (maybe a raspberry pi) to grab the data and publish the telemetry to the graphical screen or to a central control point via LoRa.

This issue document the exchanges from the F411RE card and the collector (the raspberry pi).

It can send 3 types of information row :

- the data snapshot, sent at each tick, so every 1/100s.
- the machine state snapshot, sent at the beginning of every respiration cycle, so about 35 times per minutes
- the alert trap, sent each time an alarm is triggered and each time the alarm disappear.

Each of theses informations schemes are defined in a proper comment.

Mefl, 2020-04-07:
The data snapshot (sent each 1/100s)

Data is composed with 9 fields, each field is written in binary to compress the data flow (as these exchange will not be used to be human readable).

- The MCU serial ID, to identify the machine (96bits)
it can be retrieved at `*(uint32_t*)(UID_BASE)` : 
`Serial.println("UID [HEX] : "+String(*(uint32_t*)(UID_BASE), HEX)+" "+String(*(uint32_t*)(UID_BASE+0x04), HEX)+" "+String(*(uint32_t*)(UID_BASE+0x08), HEX));`

- the Systick of the STM32 (64bits)
- centile number (16bits)
- current pressure (16bits)
- phase and sub-phase number encoded as it :
* Inspiration - inspi = 17
* Inspiration - hold = 18
* Expiration - expi = 68
* Expiration - hold = 72
So the phase is coded on the 4 high bits, and the subphase is coded on least 4 bits

- blower valve position (8 bits)
- patient valva position (8 bits)
- blower rpm (8 bits)

- battery level (8 bits)

all theses fields are separed by <tab> char.

Mefl, 2020-04-07:
alarm Trap 

- The MCU serial ID, to identify the machine (96bits)

- the Systick of the STM32 (64bits)

- centile number (16bits)

- current pressure (16bits)

- phase and sub-phase number

- the cycle number (32 bits)

- the alarm code
- the alarm level 

- triggered / untriggered toggle encoded as it :

> if triggered => the 4 high bits are HIGH and the 4 least are LOW (11110000)
> if suppression msg => the 4 high bits are LOW and the 4 least are HIGH (00001111)

- wanted value
- measured value 

- duration of the alarm in cycle count (0 for trigger message)

fpistm, 2020-04-08:
> * The MCU serial ID, to identify the machine (96bits)
>   it can be retrieved at `*(uint32_t*)(UID_BASE)` :
>   `Serial.println("UID [HEX] : "+String(*(uint32_t*)(UID_BASE), HEX)+" "+String(*(uint32_t*)(UID_BASE+0x04), HEX)+" "+String(*(uint32_t*)(UID_BASE+0x08), HEX));`

FYI, one other way could be to use LL:
```
#include "stm32yyxx_ll_utils.h"
...
  Serial.printf("%x%x%x\n", LL_GetUID_Word0(), LL_GetUID_Word1(), LL_GetUID_Word2());
```

Mefl, 2020-04-08:
all the pressure values are in mmH2O, so the typical plateau is around 300, and the peep around 80

ThomasL01, 2020-04-10:
Hello, I have started a PR for this topic #141
Feel free to comment and propose solution and idea.

pi-r-p, 2020-04-11:
A quick look at UART performance, with HardwareSerial lib : 
- There is no DMA implemented
- There is no hardware FIFO (because the CPU is DMA capable...)
- There is a 64 bytes software FIFO. It means an interrupt is triggered after each byte confirmed sent to push the next byte into TX register.

At 115200 baud:
- Transmission of 1 to 62 bytes take 50µs CPU time when calling Serial6.print (constant time)
- If you want to send X bytes with X greater than 62 bytes, calling Serial6.print takes 50µs+(X-62)*95µs (mean, see measures)

Measures at 115200 bauds:

| byte count | blocking time | time on tx line |
|--|--|--|
| 1 | 50 µs | |
| 62 | 50µs | |
| 128 | 6.1ms | 12.1 ms |
| 256 | 18.2ms | 24.2ms |

Effective max bandwith with the interrupt overhead is  10.5 ko/s (without any other preemptive interrupts)

fpistm, 2020-04-12:
Buffer size can be increased:
https://github.com/stm32duino/Arduino_Core_STM32/blob/61b14ecb4ce640bbb37b733791b51aa28d906e56/cores/arduino/HardwareSerial.h#L43-L48

IRQ prio is 1 so not advised to increase to 0 as 0 is used for the systick : 
https://github.com/stm32duino/Arduino_Core_STM32/blob/61b14ecb4ce640bbb37b733791b51aa28d906e56/cores/arduino/stm32/uart.h#L53

pi-r-p, 2020-04-12:
Not easily... I tried  `#define SERIAL_TX_BUFFER_SIZE 200` before including anything else, but no way to compile in the order I want.

fpistm, 2020-04-12:
Use this:
https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

dsferruzza, 2020-04-12:
@fpistm How does it work with Arduino CLI?

It seems I already have `build.xSerial=-DHAL_UART_MODULE_ENABLED` in my `Arduino15/packages/STM32/hardware/stm32/1.8.0/platform.txt` file. If I want to add another option like `-DSERIAL_RX_BUFFER_SIZE`, where should I put it?

fpistm, 2020-04-12:
Add a file named `build_opt.h` in your 'sketch' folder, which is (if I'm not wrong) : `src/software/firmware/srcs`

Then in this file add `-DSERIAL_TX_BUFFER_SIZE 256`
Note that I used a  "power of 2" buffer size which is recommended for ease modulo operation.

fpistm, 2020-04-12:
I've tested and redefinition works as expected.
I've made a mistake in my previous post to the content, I've forgot an '=', I've edited my comment.

pi-r-p, 2020-04-12:
Tested, it works. Nice. 
That's kind of weird to have a header file in the srcs directory... But it is working. Don't forget to `rm -rf /tmp/arduino-*` before compiling again.

fpistm, 2020-04-12:
I know .h is not a good extension but Arduino IDE could only  create file with .h, .hpp, .c, .cpp and .ino extension. That's why .h is used.
For the clean you could use arduino-cli cache clean before the first build with a build_opt.h file

valeriansaliou, 2020-04-14:
As a note discussed w/ @dsferruzza this morning: we should go for a pure-binary protocol, where the receiver parses the packets by simple binary array dereferencing. The first 8 bits should contain the protocol version (eg. V1 as 00000000, V2 as 00000001, etc.). This way the receiver can handle version changes smoothly and support multiple firmware / board versions.

nimamoslemy, 2020-05-03:
Hi, we are a team that needs your feedback to complete our idea.
Please visit our pages and comment on it. 
[github](https://github.com/nimamoslemy/ventilator)
[gitlab](https://gitlab.com/nimamoslemy/avp-ventilator)

closed 2020-05-16 by dsferruzza