makers-for-life/makair #205: Use of pneumatic air line to replace the blower.

bruprescott, 2020-04-26:
Is it possible to use a pneumatic air line instead of the blower?

Here in Brazil, all campaign beds are equipped with hospital pneumatic air and oxygen lines.
Could you add a pressure sensor to guarantee the maximum input value?

valeriansaliou, 2020-04-29:
Hello!

We already have a pressure sensor in the system, which helps perform a PID to adjust the blower speed (through the pinch valves) in order to reach the target pressure.

That means that blower speed is static in practice, and that airflow pressure is managed solely through the pinch / pressure valves. Thus, removing the blower and connecting directly on an external pressure system could work for you, in theory.

bruprescott, 2020-04-29:
Hello Valeriansaliou,
I analyzes the code and wanted to confirm if it was possible, I will try to reproduce it here and return with the analysis.  I‘m a volunteer here in Brazil to contribute to the fight against the covid-19.  Our health care system is collapsing and I‘m willing to use all the resources of my own company to help save lives.  
Doubts: Your datasheet does not contain the values ​​of the integrated circuits, but from the photos I noticed that they used drivers for the pressure valve servos.  It is not clear how the hardware will be managed.  I saw your v7 valve that uses a brushed motor with planetary reduction and encoder.  But in the code I did not identify the input for managing the encoder.  Regarding the sensor, which model do you use differential pressure sensor?  (MPX5xxx) or Honeywell?  I saw that it also includes input for flow sensor but it is not soldered on the board, will it be implemented in the future to measure the flow?  Could you help me, so that I can replicate some of the hardware for testing?

valeriansaliou, 2020-04-30:
@pi-r-p ; can you help on this? Not sure I can answer on those specific questions.

pi-r-p, 2020-04-30:
Hello, 
We play with the blower speed to avoid a huge max pressure, and the pinch valve may not support the 4bars of a pneumatic outlet. You will need something to get 70cmH2O from the mural outlet if you don't want to blow lungs up.

Indeed, we masked the sensor references in the schematic because the sensor market is really complicated right now. Everyone wants pressure sensor. You can take anything that can measure up to 80cmH2O. 
In the v7, we use a brushless motor with a reduction. There is a lot of electronics included in the motor itself that allow auto homing (find the zero index with an external optical sensor), then you can directly send a PWM as an angle. Nice piece of tech, but also hard to find on the market. I am currently working on a steppers motors version, easier to source.

bruprescott, 2020-05-01:
My idea is to use an inlet pressure regulator, a solenoid to stop the compressed air and a safety relief valve. A pressure sensor must monitor the input line to ensure that everything is within the parameters.

Regarding the pressure sensors, I have enough quantity to meet these specifications.
I don't have a flow sensor, mouser.com has a stock price around 120 USD.

I know these engines, I've used similar ones in robotic submarine projects.
Stepper motors are easy to purchase, low cost, good torque and easy to implement. The best substitute for this application.

About the hardware: could you provide me with an ics reference?
U1, U5, U11, U17, U18, U19, STM NUCLEO (32F303 or 2F411?)

About the firmware:
Is the weaning function expected for subsequent updates?
SIMV, SBT?

bruprescott, 2020-05-02:
@valeriansaliou 

Can these questions be answered publicly or only after the project is fully consolidated?

nimamoslemy, 2020-05-03:
Hi, we are a team that needs your feedback to complete our idea.
Please visit our pages and comment on it. 
[github](https://github.com/nimamoslemy/ventilator)
[gitlab](https://gitlab.com/nimamoslemy/avp-ventilator)

bruprescott, 2020-05-04:
Hello,

I managed to assemble the entire circuit, with step down converters, and more channels for pressure sensors.
I found SMI sensors with higher precision than MPX, with a close cost. It uses i2C 16bit resolution.

My idea is to use a Venturi tube and differential pressure sensor to measure the inspiration and expiration flow.

Could you help me with the implementation of PRVC, ACV controls?

valeriansaliou, 2020-05-04:
@pi-r-p :)

pi-r-p, 2020-05-04:
Hi, 
Problem with the mass flow meter is global sensor availability. We tried venturi tube, mass flow meters, and we still struggle on differential pressure sensor availability. Sensirion is a good example of overstressed company right now. You have to place an order with 40% prepayment without any delivery delay, then wait for their feedback to have a delay... within 12 weeks.
If I look at SMI SM9233 sensor (perfect for a venturi), there is nothing available on the market.

PRVC is not yet implemented, we currently work with pressure only. Volume is a great indication for obstructive cases, so we need to add it, and we are working on it.

bruprescott, 2020-05-05:
Hi @pi-r-p 

I got the following models of differential sensor in stock:
SM9333
SM9541
SM7321

The model SM9233 (Gauge Pressure Digital Sensor, 0 to +250 Pa) has 98 units in stock.

Hi @p,

I got the following models of differential sensor in stock:
SM9333
SM9541
SM7321

The model SM9233 (Gauge Pressure Digital Sensor, 0 to +250 Pa) has 98 units in stock.

I added two drivers for stepper motors, and input for optical sensors.

closed 2020-05-06 by bruprescott