makers-for-life/makair #207: Initial contribution of a system design

ejuliot, 2020-04-27:
Hello Makair team,

We tried to help you by creating a system design of the MakAir project.
The goals are to provide:
- a functional analysis to explain what funtion each part has to perform
- an architecture schema to explain how each parts of MakAir interacts with the others

By providing a visual system view of Makair, it become easier to ramp up on the project as contributors can have an overview of the system without going to deeply in implementation considerations (like electronics, software, mechanic). A diagram can be easier to read than long document. For certification and validation, it can also be helpful as it provides a proof of consistency between Hardware, Software and Specifications.

It is available here:
https://github.com/ejuliot/makair/tree/master/src/system/makair

Thanks to these diagrams, we already detected some possible improvments to the MakAir documentation. Of course, it is an initial contribution and other diagrams / analysis could be created depending on project's need (state machine, data flow, system interface description, ...) . Any initial feedback is welcome.

Etienne, Marceau, Léonie.

baptistejamin, 2020-04-28:
Hello Etienne, Marceau and Léonie for this first contribution.

If you want to, you can join the Makers For Life project, in your Slack Team.

Finally, we could send you some initial 3D parts of the first Makair machines :)

valeriansaliou, 2020-04-29:
(PR manually merged)

ejuliot, 2020-04-29:
Hi,

OK. Can you add me to the slack team ? etienne.juliot@obeo.fr (we will 
use only one account for now).

*Etienne JULIOT*
Vice President
+33 2 51 13 55 94 | +33 6 86 78 14 82
@ejuliot

<https://www.obeo.fr/>

7 Boulevard Ampère - Carquefou - France
*obeo.fr* <https://www.obeo.fr/> | *twitter* 
<https://twitter.com/ejuliot> | *linkedin* 
<https://www.linkedin.com/in/etiennejuliot>

eliottvincent, 2020-04-29:
Hello there, I just sent you an invitation.

closed 2020-05-08 by eliottvincent