makers-for-life/makair #224: Control: Graph rendering issues

BlackYoup, 2020-04-30:
So I've be digging a bit into the graph rendering problems. There are mostly 3 issues:

- We render more pixels than the chart's drawing area has. Currently the chart drawing area is ~730px large but we store ~900 points (100 points per second * 9). Which makes 1.2 points per pixel. This might be an issue (see https://github.com/38/plotters/issues/136#issuecomment-621292991). You can get the drawing area using https://docs.rs/plotters/0.2.12/plotters/chart/struct.ChartContext.html#method.plotting_area

- We receive the pressure through the Telemetry in the `mmH2O` unit and we then render the graph using the `cmH20` unit. We currently divide the pressure using a simple `/ 10` which is wrong because it drops the decimal values and make the graph renders approximate values, creating a "stairs" effect. We could instead try to render by keeping the decimals. I tried that and it makes the graph render a bit better and smoother but another problem arises: because the values move a lot (127, 128, 129, 128, ...), the rendered line won't be "flat" even though the values are very close. Instead, we will be able to see multiple dots, one for each value. I only tried with a stroke_width of 1, maybe it's better with a stroke_width of 2 but I don't think so.

- The pressure values received from the telemetry may vary. Some values go a bit too high (or too low) compared to the graph's "trajectory". The issue is way more visible with a stroke_width of 2 than it is with a stroke_width of 1. This causes some sort of "spikes" rendered.

For example, here are some values received from the telemetry, extracted from the "long" recording:
```
2020-04-29T23:50:43.500160221Z: 238,
2020-04-29T23:50:43.490150221Z: 247,
2020-04-29T23:50:43.480140221Z: 238,
2020-04-29T23:50:43.470129221Z: 241,
2020-04-29T23:50:43.460119221Z: 238,
```

The variation is quite huge and it becomes way harder to draw a smooth graph line with those. I'm pretty sure that's normal behaviour but should we try to sleek them?

cc @valeriansaliou @dsferruzza

valeriansaliou, 2020-04-30:
Thanks for the details; I've tried doing a quick native display in mmH20 and the graph indeed looked much smoother. I have a ToDo to implement a permanent, clean fix for that on next Friday.

baptistejamin, 2020-05-02:
Your master solved the problem using a low pass filter.

closed 2020-05-02 by baptistejamin