makers-for-life/makair #230: Question about UI design

FabriceSalvaire, 2020-05-02:
I just looked the "very unusual and complicated" (*) code of the control UI. ( * in comparison to many UI frameworks used in the industry )

I wonder why you didn't use the framework https://www.qt.io/qt-in-medical instead. For example, this medical device https://www.medec-intl.com/products/anaesthesia/caelus/ uses Qt for its touch user interface.  It is a good point that a framework like Qt has embeded industrial implementations.

Rust is an interesting new language, like many other experimental languages but for the UI of a medical device ???

Moreover the fact the UI system requires a X11 server on a Raspberry Debian looks funny for a true medical device.  Qt can run on many platforms and in particular on bare metal.

waxzce, 2020-05-02:
Hi

We choose the raspi essentially for a stock availability perspective, and since the UI part is not on the critical path but essentially listening logs, it can be a Linux without very big problems, even if it's suboptimal. 

The choice of rust is more for the security and resiliency it allows to use, but as you can see, the project is modular, so another UI can be written, and in the future we can remove the raspi for something smaller if we find some availability.

So, feel free to create another UI, but I will not go for the eternal language/framework debate here ;-).
In software like in every part of the project, we do favorite multiple solutions and parallel experimentations.

waxzce, 2020-05-02:
I close the issue, reopen if you need a repo for your QT UI

closed 2020-05-02 by waxzce