makers-for-life/makair #241: Blower motor details

ilyautov, 2020-07-27:
Hi!.
We started research and development for build engine for blower. We found engineer who can made that.
We decided to pay in full for this work and transfer that to the Makair project as our contribution to the common cause.
For design this we need the parameters of the engine. We need the maximum and minimum engine rotatios per minute, as well as their dependence on the control signal from the board (the so-called timing diagram).
https://blog.avislab.com/uploads/2013/05/brushless3_4.png

ilyautov, 2020-07-27:
@pi-r-p @valeriansaliou can you help us?

pi-r-p, 2020-07-27:
Hi, you plan to build a brushless motor on your own ?
The main blower challenge is not really the motor, but the blower itself, with a perfect balance to spin at full speed without too much vibration.
The challenge is the motor+rotor and turbine.

Globally, the blower must achieve this:
[image]

or this (red curve for 24V)
[image]

Current blower is around 40000rpm
If you can achieve this with less than 40000rpm, that could be great too.

Next issue is the noise... It is highly linked to the blower design. What we saw in medical blowers we teared down is that it is common to sacrifice efficiency to avoid high frequency noise.

Timing diagram is not that much important compared to other blower challenges. We also used sensorless ESC that did a really good job (drone ESC).

ilyautov, 2020-07-29:
Hi, @pi-r-p thank you for your response that is will be helpful for build.
Yes, i think we can try to design full set of motor, rotor and turbine.
Can we chat about that directly? for example in telegram or email: @wh_zt ilyautov@gmail.com
Can you advice to us any motor which we can use with blower v14 for start redesign?  F.e: Parrot Bebop Drone 2 Motor Kit or Parrot Anafi motor kit?